thermofisherlsms/iapi #2: 64 bit support

rfellers, 2016-09-02:
Our application requires the ability to cache more than the 32 bit .NET limit in order to speed up data processing during acquisition.

dbaileychess, 2016-10-27:
This is still in discussion, as our main Tune application is not 64-bit enabled at the moment, the API is also forced to be 32-bit.

massfrontier, 2016-12-06:
Any updates on this feature?

dbaileychess, 2018-04-04:
We don't have any immediate plans to support a native 64-bit. I will be moving this to Future and will update accordingly.

joewandy, 2022-07-19:
Hi @dbaileychess thank you for your hard work on IAPI! 

I wonder if there has been any update on this issue since 2018?

jesse-canterbury, 2022-07-19:
Hi, thanks for bumping this issue!  The version of Fusion.API-1.0.dll that's currently uploaded here was built for the "Any CPU" configuration, so it should work with your 64-bit applications.  If you need one that's built explicitly under the "x64" configuration, please let us know.

jesse-canterbury, 2022-07-22:
I'm going to close this issue now, as 64-bit functionality should now be supported.  If anyone continues to have problems of this sort, please open a new issue.  Thanks!

closed 2022-07-22 by jesse-canterbury