thermofisherlsms/iapi #4: Event for Contact Closure

rfellers, 2016-09-02:
We would like to respond to contact closure events (from our LC) through the IAPI

dbaileychess, 2016-09-02:
So when an external device sends a contact closure, you would like to receive that through the API?

rfellers, 2016-09-02:
Ya, we'd like to start and stop our acquisition based on the contact closure.  Please let me know if this is crazy talk ...

dbaileychess, 2016-11-04:
If you are going to run a method, there is now an option to have it wait for contact closure, just like you would from Tune/Xcalibur. Will this suffice for now, or do you want a dedicated event?

rfellers, 2016-11-04:
I'm not sure if that will work with our current code.  We do everything through IScans.CreateCustomScan() as each scan could potentially be unique.  Is it possible to wrap that in a method?

dbaileychess, 2016-11-04:
I guess I misspoke above. Any acquisition, either by method or just time/scans/continuous, has the option to be started on a contact closure event. 

I'll still explore if we can implement a dedicated event for it, as it may have other uses in the future.

commit 099de9a4 (2016-11-08): Added Contact Closure support for API. Closes #4
closed 2016-11-08 by dbaileychess