thermofisherlsms/iapi #10: The instrument version (3.1) does not support the API

massfrontier, 2017-11-11:
Hi,
We have installed Orbitrap Fusion 3.1.2162.0. Unfortunately we are getting exception "The instrument version (3.1) does not support the API".

Is there anything we can do to fix this?

Thanks

jacobrayschwartz, 2017-11-13:
Hi,

Have you installed the API license for 3.1? This is done via Tune and is required before being able to use the API.

Jacob

massfrontier, 2017-11-13:
Hi Jacob
We believe we have license applied properly
[image]

jacobrayschwartz, 2017-11-13:
Ah, I see now. There was an issue in that release that we didn't see due to development environments. It's been fixed in the latest build. Sorry about the confusion. I'll get the build uploaded for you.

closed 2017-11-13 by jacobrayschwartz