thermofisherlsms/iapi #11: q-exactive support

trishorts, 2018-01-16:
what is the status of this?

dbaileychess, 2018-02-06:
We are looking into possible solutions of hosting the QE IAPI either in this repo or in its own. We have not finalized that decision yet, but I will keep this issue updated on those developments.

trishorts, 2018-02-06:
thanks Derek. really looking forward to this.

tobiasko, 2018-06-21:
Hi Derek,

following your ASMS2018 talk entitled:

"iAPI Development and XML Modification Overview: Thermo Scientific™ Tribrid™ and Exactive™ MS Series"

What is the current situation? Is there a separate iAPI for FUSION and Exactive platforms? Or is this repo now the main entry point for both systems? The current situation is pretty confusing.

Greetings,
Tobi

dbaileychess, 2018-06-22:
Tobi,

There is a functionally equivalent, binary-incompatible, version of the iAPI for the Fusion-series instruments. Both APIs are now hosted in this repo. We kept the two APIs in one repo since they do share so much in common that a shared issue-tracker and documentation made sense. Directories are split between Exactive-Series and Tribrid-Series files.

I will be adding more documentation in the upcoming weeks to help clarify the distinction. Please let me know of any specifics on the confusion and I will attempt to address them.

tobiasko, 2018-06-28:
Hi Derek,

the mayor point of confusion is currently the licence agreement.  There is one agreement for all platforms?

Greetings,
Tobi

tobiasko, 2018-06-28:
Hi Derek,

which Tune version is needed on the Exactive platforms?

Greetings,
Tobi

closed 2023-05-09 by jesse-canterbury