thermofisherlsms/iapi #19: Question - Which field in the Trailer corresponds to the RunningNumber in IScanDefinition?

chhh, 2018-07-13:
Which field in the Trailer corresponds to the RunningNumber in IScanDefinition?

dbaileychess, 2018-08-21:
The field would be the 'Access ID:' field in the trailer-extra portion of the rawfile. There is some historical reasons for the name change and we didn't feel like it was necessary to unify the name at this time. Perhaps in the future the property and field in the rawfile will be common.

closed 2018-08-21 by dbaileychess

chhh, 2018-08-21:
@dbaileychess 
Is this field available during online acquisition with the IAPI? I couldn't find a field named "Access ID" in the Trailer of an arriving IMsScan. 

Here's the code I use to dump Trailer information from `IMsScan` (using `scan.Trailer` as `source` argument):
```csharp
public static string InfoSourceToString(IInformationSourceAccess source, bool outputEmpty)
{
    if (source == null) return "NULL";
    if (!source.Available) return "Not available";
    if (!source.Valid) return "Not valid";
    var entries = new List<string>();
    foreach (string key in source.ItemNames)
    {
        if (source.TryGetValue(key, out var val))
            entries.Add($"{key}: {val}");
        else if (outputEmpty)
            entries.Add($"{key}: N/A");
    }

    return entries.Count == 0 ? "No entries" : string.Join(", ", entries);
}
```