thermofisherlsms/iapi #21: Contact closure event for Exactive

chhh, 2019-11-07:
Looks like ContactClosure event is only found in Fusion.

jesseatthermo, 2019-11-22:
That event is not currently defined for the Exactive API, true, but the following events should offer you some options for implementing what you want:

+ The AcquisitionStreamOpening event is fired, if the acquisition is started after a contact closure is triggered, when toggled as starting criteria.
+ The StateChanged event will be fired, if the SystemState is changing from the “Waiting for contact closure” state to another one.
+ An Acquisition can be created via API and a ContactClosure can be defined as trigger (IAcquisitionWorkflow.Trigger), to wait for.

closed 2019-11-22 by jesseatthermo

chhh, 2019-11-22:
Oh, that last one is useful, thank you!