thermofisherlsms/iapi #22: CanAcceptNextCustomScan Event is never raised

caetera, 2020-08-18:
After subscribing for CanAcceptNextCustomScan event of IScan interface the action is never invoked. I have also tested the provided example tool (FusionExampleClient) with the same result – i.e. the customScans counter never got incremented. Was this event deprecated and if so what would be the recommended way to find out when to send a custom scan to the instrument?
Tests were performed both on Lumos and Eclipse systems, both running Tune 3.3 (if it is important).

jesse-canterbury, 2020-08-25:
Thanks for the report.  I'm looking into this and should have a response soon.

jesse-canterbury, 2020-08-25:
Thanks again for the report.  I can confirm that the problem is there & will add the issue to the bug list.  Unfortunately this will not be fixed in time to be included in Tune 3.4.

closed 2020-08-25 by jesse-canterbury

caetera, 2020-08-25:
Thank you for the answer. What would be the "correct" way to determine if a new CustomScan can be sent to the instrument?

dbaileychess, 2020-08-25:
New custom scans can always be "sent" to the instrument, the instrument will enqueue them if it has space (I forget the size of the queue, but it is 10-100 in size about, @jesse-canterbury could find the actual number). If it doesn't have the space, it will overwrite the oldest scan in the queue (i.e., it uses a circular buffer). I don't think there is user-visible message if that occurs, I know I had a log print in there to log when that happens.

The CanAcceptNextCustomScan event is used when the instrument dequeues a custom scan and is about to execute it. It just a feedback mechanism to alert the API that it has space for new scans.

I'm not familiar why the event isn't firing at the moment, Jesse will help look into that.

The user of the API should try to pace the Custom Scans it sends itself, using that event to help it decide. So it would be safe for now (until the bug is fixed) to keep track of how many Custom Scans you send, and how many you see are executed by monitoring the scans that were executed.

caetera, 2020-08-26:
Thank you for explanation, @dbaileychess