thermofisherlsms/iapi #23: No prescans when instrument under API control

caetera, 2020-08-18:
As soon as the instrument is controlled by the IAPI no prescans are performed any more. The issue appear both if a repeating scan is scheduled or a series of custom scans. As soon as the control returned to Xcalibur (for example, by canceling the repeating scan) the prescans appear as usual. The trailer reports AGC as `Predicted`, `On (Tradition)`, or `On`, but the injection time is always *exactly* the same (to the last visible digit) as it was in the last "non-API" scan. The following error is reported in the instrument console.
 ```
Failed to copy intermediate trailer data to byte array :
	IntermediateStorage : failure to find name 'masterScanNumber'
```
Tests were performed both on Lumos and Eclipse systems, both running Tune 3.3

jesse-canterbury, 2020-08-25:
Thanks for the report.  This is actually performing as expected.  The workaround for the prescan issue is to briefly return to normal method control (i.e., no API scans) to allow a prescan to be performed.  Then you can resume your API scans.  Note:  we have recently implemented a way to do AGC under API control.  This will be made available as part of Tune 3.4, to be released soon.  

The trailer issue you mentioned is related, but more of an annoying symptom rather than pointing to a cause.  

Thanks again for the submission.

closed 2020-08-25 by jesse-canterbury

caetera, 2020-08-25:
Ok. Thank you.