thermofisherlsms/iapi #25: CanAcceptNextCustomScan is not raised after upgrading to Tune 3.3

joewandy, 2020-11-23:
We were using the older Fusion API (3.0.1794) with Tune 3.2, and it was working fine. Recently we were upgraded to Tune 3.3, and I notice now that the `CanAcceptNextCustomScan` is not raised when a new scan has arrived. 

I also did a test using the Fusion example client by adding some console print at [this line](https://github.com/thermofisherlsms/iapi/blob/master/examples/tribrid/fusion-example-client/Form1.cs#L222) that handles the canAccept event, and nothing was printed. Before this code used to work fine without any problem on Tune 3.2. Do you have any idea what we could do to resolve this?

jesse-canterbury, 2020-11-23:
This is actually a known issue and will be fixed in the next version.  In the meantime, you can read the comment thread for issue #22:  https://github.com/thermofisherlsms/iapi/issues/22

In that thread there is an explanation for how that event works.  The short answer is, it's not strictly necessary to monitor for that event.

joewandy, 2020-11-24:
Thanks for the quick reply @jesse-canterbury. I'll implement the suggestion in that thread to 'keep track of how many Custom Scans you send, and how many you see are executed by monitoring the scans that were executed'.

I'm also wondering, when you say that it will be fixed in the next version, did you mean in Tune 3.4? We couldn't upgrade to that yet because the PC connected to the instrument is not running Windows 10, but eventually we might do that.

closed 2020-11-24 by jesse-canterbury