thermofisherlsms/iapi #26: Fusion iAPI inclusion/exclusion list not yet implemented

im281, 2021-01-03:
`ITable table;
 ITableRow row;
 table = instrument.Control.Methods.CreateTable(typeof(IExclusionTable));`

returns 'not yet implemented' 

Do I need to implement the IExclusionTable interfece? For the the Exactive series it is a different method signature
`       ITable table;
            ITableRow row;
            table = instrument.Control.Methods.LoadTable(methodPath, 1, typeof(IExclusionTable));`

jesse-canterbury, 2021-01-05:
Tables aren't supported for the Fusion series at present (but should be working for the Exactive series).  We're looking into adding this support for the next Tune release.

jesse-canterbury, 2022-07-07:
In the works, yes.  But unfortunately I can't give you a firm timeline, sorry.