thermofisherlsms/iapi #28: tribrid method-filter example not updated?

jessicawang025, 2021-06-21:
AcquisitionStreamOpening and AcquisitionStreamClosing are not defined in IFusionMsScanContainer in the **tribrid method-filter example** (errors in lines 107, 108). 

MsAcquisitionOpeningEventArgs also couldn't be found (line 116).

jesse-canterbury, 2021-06-21:
You are correct, this example has not been updated in some time.  I'll be revising the IAPI examples in the coming weeks in order to align with more recent developments.  Hopefully after that this example will work for you.  I'll leave this open until then.

jesse-canterbury, 2021-07-07:
Greetings again.  It appears these build errors are easily corrected.  MsAcquisitionOpeningEventArgs should be changed to AcquisitionOpeningEventArgs (i.e. remove the first two letters).  Similarly, AcquisitionStreamOpening and AcquisitionStreamClosing are defined in the InstAcq, so you can change it to this:

[image]

This should allow the project to build.  I will be updating the examples soon with these corrections, but hopefully this will allow you to get going.

jesse-canterbury, 2023-05-09:
Closing this now since the original issue was fixed.

closed 2023-05-09 by jesse-canterbury