thermofisherlsms/iapi #30: AGC and MaxIT not working for Orbitrap MS2 scans on Eclipse

pisistrato, 2022-07-30:
As can be seen in the gif, when trying to replicate a MS2 scan on the Eclipse using the provided `fusion-example-client` example neither the AGC nor MaxIT are taken when placing the scan from the API

[image]

When the readout is set to the IonTrap it works as expected.

jesse-canterbury, 2022-08-01:
Thanks for the report.  Have you installed the patch for 3.5?  You should be able to get to it [here](https://github.com/thermofisherlsms/iapi/tree/master/misc).

That said, I can see the issue on my side as well.  However it works as expected when running a method acquisition "underneath" the IAPI application.  I'll look into it.

pisistrato, 2022-08-01:
Yes, the instrument is patched.

Indeed, the issue appears only when the scan is placed "directly" from the IAPI.

Thanks.

t0mdavid-m, 2024-01-12:
We are having a similar issue with Tune 4.0. Changing the `AGCTarget` has no effect on Ion Injection Time. We tried varying `AGCTarget` between `3000` (reported default) and `100000` (maximum valid value).

Therefore, I assume that this parameter is currently not working and some default is used instead. However, given that according to the tune documentation 100% normalized base value is equivalent to `400000` and `50000` for Full and MSn scans respectively (Orbitrap), the documented value range and default value are unexpected.

In order to ensure comparability between Tune and the iAPI I would like to enquire what AGC target values are used internally when measuring via the Orbitrap (i.e. what is the input that should be made to Tune in order to obtain an equivalent AGCtarget)?

Platform: Orbitrap Fusion Lumos Tribrid Mass Spectrometer + FAIMS Pro Interface

jesse-canterbury, 2024-01-12:
Are you placing scans while running "on top" of Tune or while running a method?  Placing scans while running on top of Tune still has problems; this issue has been deprioritized as the vast majority of users use the IAPI in conjunction with methods.  

And what injection times are you seeing?  Are they all the same, i.e., set to the maximum injection time?

t0mdavid-m, 2024-01-15:
Thank you for the swift response!

We tried adjusting the `AGCtarget` on top of a Method and still does not seem to be working. 

Setting `AGCtarget` to `3000` or `100000` did not change the IIT for both the iontrap and orbitrap mass analyzers. The IIT was also smaller than the maximum injection time (~50ms for FTMS with maxIIT=100ms).

jesse-canterbury, 2024-01-15:
Can you attach your method here, so that I can try it as well?

t0mdavid-m, 2024-01-16:
Below is the method we used:

[method.zip](https://github.com/thermofisherlsms/iapi/files/13946999/method.zip)

Over that method we used the `fusion-example-client` to trigger scans via the iAPI.

jesse-canterbury, 2024-01-24:
Apologies for the delay.  This appears to be related to how the fusion-example-client sends scans.  It is sending scans with all possible values defined; normally one would make a scan with only the relevant parameters defined.  

If you comment-out line 322 in Form1.cs of FusionExampleClient, and instead set the values yourself, for example using:

```
cs.Values["FirstMass"] = 400.ToString();
cs.Values["LastMass"] = 1400.ToString();
cs.Values["ScanType"] = "Full";
cs.Values["Analyzer"] = "Orbitrap";
cs.Values["ScanDescription"] = "Full400-1400";
cs.Values["AGCTarget"] = 5.ToString();
```

...and then for "AGCTarget" try different values (switch between 5 and 500,000 to see huge differences), you should see the injection times varying as expected.  

Hopefully that will get you going.  Probably there is still an issue here, which I'll continue investigating.  Thanks again for the report.

jesse-canterbury, 2024-03-25:
OK, after returning to this issue recently, and with the help of another user, we found that AGCTarget is working as intended for full scans, but other scan types appear to have a problem.  The workaround is to use instead the "MSXTargets" parameter, like so:

`cs.Values["MSXTargets"] = "10000";`

And then continue using "AGCTarget" for full scans.  This will be fixed in Tribrid series 4.2, due out this summer.

pisistrato, 2024-03-27:
Thanks for the feedback, I can confirm this workaround works fine.

@jesse-canterbury off-topic question: to whom can I report a bug I discovered on the Exploris tune? It concerns a charge state decision tree DDA method. The method is compiled (i.e. it is saved as .meth), but it is not executed (i.e. an error appears when trying to submit the queue)

jesse-canterbury, 2024-03-27:
@pisistrato I forwarded your note to the relevant folks on the Exploris team.  They'll take a look & make a bug report.  If I get any more info back, I'll update you.  And feel free to ping me if enough time goes by with no news.  

Glad to hear the workaround above works ok!

jesse-canterbury, 2024-03-27:
@pisistrato Please email [support.exploris.bremen@thermofisher.com](mailto:support.exploris.bremen@thermofisher.com) with a full set of info about your observations, such as Exploris software version, your method file, the displayed error message, and steps to reproduce the error.  And please let me know if I can help at all.  Thanks!

skaplich, 2026-09-29:
Hello, wanted to reopen the issue, running simple client to submit full MS2 scans.

**Platform:** Orbitrap Fusion Lumos Tribrid  
**Software version:** 4.3.4477.11 (post-4.2, where this was reported as fixed)  
**Acquisition mode:** On top of Tune (no method file)

Adjusting AGCTarget on scans submitted from IAPI (on top of Thermo Tune) does not seem to have any meaningful effect on injection time. The MSXTargets workaround from #30 partially works, but its behavior depends unexpectedly on the number of comma-separated values provided.

This is what the expected scan parameters would be using MSXTargets: 

cs.Values["MSXTargets"]      = "10000000”;

Injection time did not change and hovered around 30-50ms.

Submitting scans using the definition below did change the injection time to 180-200ms, the spectra also seemed to be more reasonable, as it matched the spectra generated when submitting scans entirely through Thermo tune (using the same parameters as IAPI).

cs.Values["MSXTargets"]      = "10000000,10000000,10000000,10000000,10000000";

```csharp
var instrument = Factory.Connect();
var scanControl = instrument.Control.GetScans(exclusive: true);

while (true)
{
    ICustomScan cs = scanControl.CreateCustomScan();

    cs.Values["MSOrder"]        = "Ms2";
    cs.Values["Analyzer"]       = "Orbitrap";
    cs.Values["PrecursorMass"]  = "1234.5678";
    cs.Values["ActivationType"] = "ETD";
    cs.Values["ReactionTime"]   = "25.0";
    cs.Values["AGCTarget"]      = "1000000";  // no effect on injection time
    cs.Values["MSXTargets"]     = "10000000"; // single value: ~30-50ms injection time
    // cs.Values["MSXTargets"]  = "10000000,10000000,10000000,10000000,10000000"; // five values: ~180-200ms
    cs.Values["MaxIT"]          = "500";

    scanControl.SetCustomScan(cs);
}
```

jesse-canterbury, 2026-09-30:
There are a few things going on here:

- "MSOrder" is not a valid key.  I think what you want is probably "ScanType" set to "MSn".  This parameter will be ignored and the scan type will default to "Full" which might not be what you want.
- If you want to use multiple comma-separated values for MSXTargets, you also need the same number of values for the PrecursorMass, otherwise the scan definition will be messed up
- As mentioned in this thread (comment from January 12 2024), placing scans "on top" of Tune, i.e. without running a method, probably will not generate expected results.  This still has not been remedied, as it has not been prioritized, because the vast majority of users are using the IAPI in the context of methods.

Also, a side note:  if you are really sending scans in an infinite loop, you will very quickly exhaust the scan queue capacity for custom scans (max is 50), and then it will ignore further scans until the queue clears some room, at which point it will fill up again.  It may be difficult to tell what's going on in this case.

Does your usage of the IAPI require running with Tune only, i.e. without running a method?