thermofisherlsms/iapi #32: pAGC for MS3 scans

jpm369, 2022-12-03:
Is there any way through to API to designate an MS2 spectrum to provide pAGC data for a set of MS3 scans, or does the IsPAGCScan bool only apply for MS1 scans? I observe that if I spin out multiple MS3s on different precursors from a single MS2 (i.e. schedule a distinct MS3 for the top 3 peaks in an MS2 spectrum), all of the resulting MS3 spectra have the exact same (not maxed) injection time independent of their MS3 precursor. Maybe more basically - what spectrum does an API inserted MS3 scan refer to for its pAGC data?

jesse-canterbury, 2022-12-03:
The scan from which pAGC data is derived must be an ion trap full scan.  If it's not, i.e. if you set the IsPAGCScan to true for anything else, you'll see a message in the console saying "WARNING:: IAPI pAGC scan definition must be a linear ion trap full scan.  Executing scan, but not as a pAGC scan."  

To answer the last question, if you set the pAGCGroupIndex for your MS3 to match the last pAGCGroupIndex for the last pAGC LIT full scan, it'll take the pAGC data from that scan.  If you want the MS3s to have different injection times, you can set the AGCTarget parameter to different values.

jpm369, 2022-12-03:
Thanks for the fast reply - just to clarify, the AGCTarget parameter is the only way to modulate the injection times of MS3 precursors that share an MS2 parent mass belonging to the same pAGCGroupIndex?

jesse-canterbury, 2022-12-03:
That's correct, but you could also do something like set a very high target value, and then set the MaxIT parameter to something relatively small.  That way you set up a situation where you expect the injection time to max out, but it'll max out at the time you specify, scan-by-scan if you want.