thermofisherlsms/iapi #33: Controlling FAIMS on Exploris

chhh, 2022-12-23:
Hi, is it possible to set Faims values on Exploris?
`InstrumentValues` says there are no commands available for this node.
I also used the example from this repository and I don't get any notifications when I change FAIMS cv from Tune. The only notification I'm getting is when I turn "FAIMS voltages" to On - then I get a notification about "Network Devices" node.

KyowonJeong, 2023-01-31:
I also wonder if FAIMS cv could be controlled via iapi in Lumos or Eclipse machine. Is it not implemented yet? Then is it planned in the future? Please let me know..

jesse-canterbury, 2023-01-31:
Yes, FAIMS can be controlled via the IAPI on tribrids.  The values to set are:  

1. "FAIMS CV" which can take values between -250 and 250V
2. "FAIMS Voltages" which can be set to either "On" or "Off"

KyowonJeong, 2023-01-31:
> Yes, FAIMS can be controlled via the IAPI on tribrids. The values to set are:
> 
> 1. "FAIMS CV" which can take values between -250 and 250V
> 2. "FAIMS Voltages" which can be set to either "On" or "Off"

That is super!!! You just opened a gate for a brand new cool project! Thanks!!

chhh, 2023-01-31:
> Yes, FAIMS can be controlled via the IAPI on tribrids. 

@jesse-canterbury Does that mean no luck for Exploris 480?

jesse-canterbury, 2023-01-31:
@chhh Sorry for the delay.  I am still waiting on information from some colleagues to address your question.

chhh, 2023-03-03:
@jesse-canterbury No luck yet?

jesse-canterbury, 2023-05-09:
@chhh Sorry again for the delay.  Sorry to tell you that FAIMS control on Exploris is not supported via the IAPI.  It may be implemented in a future version, but no guarantees.

closed 2023-05-09 by jesse-canterbury