thermofisherlsms/iapi #34: APD is not performed when MS1 scan is placed via the IAPI

pisistrato, 2023-02-01:
I have noticed that if the MS1 scan in placed via the IAPI, the returning spectrum is only centroided, the ICentroid.IsMonoisotopic flag is never TRUE.
I was wondering if this is intended and whether it would be possible to apply APD also on custom scans in the future.

Tested on both Eclipse and Exploris, running the latest Tune versions available as of today.

jesse-canterbury, 2023-02-01:
This was [reported previously](https://github.com/thermofisherlsms/iapi/issues/17) as well.  I happy to say it's fixed in the latest Tune version 4.0 for tribrids.  I will try to get some clarity on the situation for Exploris.

pisistrato, 2023-02-01:
I was not sure the two things were related, not much info were provided in the other open issue :)
I just saw 4.2 is out (we updated to 4.1 in November). If in 4.2 that is fixed I will install it and give it a try, please let me know when you know more.
Thanks

pisistrato, 2023-07-10:
Hi @jesse-canterbury, is there any update on this for the Exploris?
I had to ask because I am working on a smart acquisition method, and although it already slightly outperforms the native DDA method, it would be nice to make a proper comparison to evaluate the full potential of my method.
Unfortunately, the monoisotopic determination algorithm I have put together does not seem to perform that well...

Thanks :crossed_fingers:

jesse-canterbury, 2023-07-10:
No updates for the moment, but a resolution will likely have to wait for a new version of Exploris control software.  I'll try to get more information.

chhh, 2023-07-13:
@pisistrato What is an APD scan?

pisistrato, 2023-07-13:
Advanced Peak Determination. Well, no monoisotopes determination is
performed at all on MS1 scans.

chhh, 2023-07-13:
@pisistrato Thank you!