thermofisherlsms/iapi #35: Is it safe to set Resolution to any value in the range returned by instrument's help?

chhh, 2023-03-23:
IAPI help for the Resolution setting on our OE480 says that it can be set to any value in the range 7500.0-480000.0.
```
Name[Resolution]
Selection[7500.0-480000.0]
DefaultValue[30000.0]
Help[Resolution setting, determines transient data acquisition duration per scan]
```
Is it sage to set Resolution to `32000.42` for example?

jesse-canterbury, 2023-03-23:
I'm not sure what validation might exist on the Exploris for making those settings.  It could be that weird values like that wouldn't be accepted at all, and might generate some sort of error.  "Under the hood," though, I believe any strange resolution settings will either be ignored or else will default to the closest valid setting (or perhaps to a default value).  In the case of 32000.42 the closest valid setting would be 30000 I think.

chhh, 2023-03-24:
Thank you for the quick response @jesse-canterbury 
I asked because I wanted to have a finer control over the duration of the Orbi scan. Up until the very recent v4.2 software for Exploris480 there was only a limited set of options. With 4.2 a bunch of extra resolution options were added (45k, 11.5k maybe some other). Would be cool if there were more options between 60k and 120k - there's a big jump from 128ms to 256ms.

closed 2023-03-24 by chhh