thermofisherlsms/iapi #38: Using the API in Python

kopysov, 2023-07-11:
I'm trying to use the API in my Python program via pywin32 package that allows accessing COM objects and calling their methods.
Similarly to "the old way" described in the Power Point presentation:
```
Type t = Type.GetTypeFromProgID("Thermo Exactive.API_Clr2_32_V1", true);
object o = Activator.CreateInstance(t);
```
in Python I instantiate the COM object:
`o = win32.client.Dispatch('Thermo Exactive.API_Clr2_32_V1')`
and this works fine. However, when I try to get an InstrumentAccess object by calling Get method of the COM object, I get the following error:
`Unable to load an assembly set being able to write RAW files`

I've tested this in both 32-bit and 64-bit builds of Python. Moreover, a C# program implementing this "old way" of accessing the API throws the same error. The new way of accessing the API via loading the .NET assembly works fine.

Do I miss anything? Maybe, "the old way" does not work anymore?

pisistrato, 2024-03-11:
I am also interested in this.
I have been experimenting with clr.AddReference() and pythonnet, I managed to get the connection to the instrument (e.g. I can DetectorClass), but I cannot get this orbitrap.MsScanArrived += Orbitrap_MsScanArrived; done in python.
It seesms the event is not recognized in python....

jesse-canterbury, 2024-03-12:
Hello!  For the question from July of last year:  this method of accessing the instrument is not supported on current instrument control software versions, sorry.  This presumably would include all Exploris versions.  Older Exactive and Q-Exactive versions may be supported; please leave a comment here indicating if that would be useful to know, and I will try to find out.

For the more recent question, I don't have experience with this.  Can you build a C# app to access the instrument and handle events, and then connect your python code to the C# app (and the event handler)?

closed 2025-02-03 by jesse-canterbury