thermofisherlsms/iapi #39: Centroid or Profile selection for placed scans

chhh, 2023-08-31:
Hi, when I connect to Exploris 480 it lists the following available scan parameters - I don't see anything for centroid/profile selection, is this right?

```text
Name[FirstMass], Selection[40.0-8000.0], DefaultValue[120.0], Help[First m/z, lower edge of displayed spectral range]
Name[LastMass], Selection[40.0-8000.0], DefaultValue[2000.0], Help[Last m/z, upper edge of displayed spectral range]
Name[IsolationRangeLow], Selection[40.0-2500.0], DefaultValue[120.0], Help[Lower m/z edge of isolation range]
Name[IsolationRangeHigh], Selection[40.0-2500.0], DefaultValue[2000.0], Help[Upper m/z edge of isolation range]
Name[MsxInjectRanges], Selection[string], DefaultValue[[]], Help[MSX m/z-ranges for isolation steps. If a single range is exceeding isolation min/max, no range is applied! Format - example:  [(191.5,194.5), (523.3,525.3), (1117.0,1127.0)]]
Name[MsxInjectTargets], Selection[string], DefaultValue[[]], Help[MSX AGC targets for injection steps. If the total sum of 10e6 is exceeded, no definition is applied! Format - example: [20000, 50000, 35000]]
Name[MsxInjectMaxITs], Selection[string], DefaultValue[[]], Help[MSX (max) inject time in milliseconds for isolation steps. If the sum of ITs is exceeding 1000ms, no defition is applied! Format - example:  [80.0, 8.5, 120.0]]
Name[MsxInjectNCEs], Selection[string], DefaultValue[[]], Help[MSX normalized collision energy for isolation steps. Format - example:  [30.0, 55.5, 17.8]]
Name[MsxInjectDirectCEs], Selection[string], DefaultValue[[]], Help[MSX HCD collision energy direct eV setting. Format - example: [10.4, 22.1, 42.8]]
Name[MaxIT], Selection[0.03-1000.00], DefaultValue[10.0], Help[Injection time in milliseconds, maximum injection time for AGC operation. If set to <=0, automatically uses the detect time of the previous scan as maximum inject time.]
Name[Resolution], Selection[7500.0-480000.0], DefaultValue[30000.0], Help[Resolution setting, determines transient data acquisition duration per scan]
Name[Polarity], Selection[0,1], DefaultValue[0], Help[Ion polarity setting, 0=positive ions, 1=negative ions]
Name[NCE], Selection[-200.0-200.0], DefaultValue[0], Help[HCD Collision Energy in normalized units.]
Name[NCE_NormCharge], Selection[1-5], DefaultValue[2], Help[ControlNode]
Name[NCE_Factors], Selection[string], DefaultValue[], Help[Factors for stepped CE for a more flexible definition. Example:  [0.01, 0.5, 0.75, 1.0] (reference is NCE/DirectCE, MAX = 1.0, best select the highest value for NCE/CE)]
Name[DirectCE], Selection[-200.0-200.0], DefaultValue[0.0], Help[HCD collision energy direct eV setting (NCE needs to be set)]
Name[SourceCID], Selection[-135.0-135.0], DefaultValue[0.0], Help[In-source fragmentation energy (eV)]
Name[Microscans], Selection[1-10], DefaultValue[1], Help[Number of microscans per logicel scan.]
Name[AGC_Target], Selection[1-10000000], DefaultValue[1000000], Help[Target number of trapped ions for automatic control (AGC). Active when mode is 'Absolute AGC Target']
Name[AGC_Mode], Selection[0,1], DefaultValue[1], Help[1=Enable automatic injection control for number of trapped ions (AGC), 0=Use fixed injection time]
Name[AGC_PeriodicPS], Selection[0.0-5.0], DefaultValue[0.66], Help[Setting this !=0 inserts wide-mass-range-AGC prescans at given period [sec] (will be active only if prescans are required by analytical scans). The period is calculated automatically when this is set to -1.]
Name[PrOSA_Support], Selection[0,1], DefaultValue[1], Help[1=Enable/0=disable PrOSA periodic operation (for AGC improvement)]
Name[Scan_Handshake_Mode], Selection[0,1,2], DefaultValue[0], Help[0=Disable(default)/1=Enable-for-all/2=Enable-for-MS1-only Peripheral connector B per-scan-handshake functionality]
Name[Scan_Handshake_MinTriggerDelay], Selection[0.0-500.0], DefaultValue[10.0], Help[Handshake min. Trigger Delay (msec). Default is 10.0 msec]
Name[Scan_Handshake_Nth_Counter], Selection[1-1000000], DefaultValue[1], Help[Setting this >1 will insert a handshake only every N-th scan]
Name[Scan_Handshake_Nth_Cnt_Cycle_Terminate], Selection[0-10000000], DefaultValue[0], Help[Setting this to a NEW value >0 will terminate and restart an active N-th-scan-conter-cycle with the next scan. Thus next scan will be handshaked even if N-th counter is not reached.]
```

jesse-canterbury, 2025-02-03:
Sorry for the lengthy delay here.  Normally centroid/profile data type would be accessed with "DataType" as the key and "Centroid" or "Profile" as the value.  Have you tried that by chance?

chhh, 2025-02-03:
No, I haven't as it wasn't listed among the supported keys when the instrument reports which values are available.
Well, I now don't have access to a 480 anymore, so I can't check.