thermofisherlsms/iapi #44: Astral support

david-bouyssie, 2024-05-29:
Is it possible to use IAPI on Astral instruments?
If not, is there any plan to support it?

Thanks.

jesse-canterbury, 2025-02-03:
Sorry for the delay in replying.  It is not currently possible to use the IAPI on Astral instruments.  There are plans to implement this, but no timeline is available yet.

closed 2025-02-03 by jesse-canterbury

david-bouyssie, 2025-02-04:
Thanks, I'm looking forward.

I understood this was not easy because of some intrinsic Astral optimizations related to ion packets management, and also the high frequency of MS/MS scans.
Naively, I guess that a different API, based for instance on batched MS/MS queries (instead of individual MS/MS queries as it is interfaced today), may give more flexibility for optimizing the internal scheduling.
But this might create other issues I'm not aware of...

chhh, 2025-02-04:
It has to be batched, otherwise the latency per request is too high, it will lower the throughput. Also, it doesn't make sense to make decisions at this time scale (single milliseconds / tens of milliseconds), you need to have your decisions about scans ready for the next second or so